opentrack/opentrack #1972: camera issues on opentrack beta 2024.1.1

sthalik, 2024-12-28:
Hey,

There are quite a few complaints that the pre-release version 2024.1.1 doesn't work with camera-based trackers. We don't have any telemetry so I need to run a poll.

Let's do the voting part using post reactions. Note that reacting to a message doesn't make the project more visible so this isn't some kind of reaction farming on my part.

When you're using a camera-based tracker such as neuralnet, PointTracker or Aruco with opentrack 2024.1.1 and:

- **if 2024.1.1 build works** for you, please react with **thumbs up**.
- it doesn't work on 2024.1.1 beta but some older version such as **2023** or even older works, please react to this post with '**thumbs down**'.
- it **doesn't work** on older versions as well (or you haven't found one that worked, or didn't look for old versions at all), react with '**rocket**'.
- if you're using an older versions and don't intend to try 2024.1.1, react with **happy face**.
- if you're not using any opentrack version, react with **eyes**.

If you want to elaborate or just comment on the vote, please *also* express your vote by reacting to this message.

Len62, 2024-12-28:
It works using the ps3eye-subprocess from an earlier version.

GO63-samara, 2024-12-29:
Works in Windows 10.
I use the driver CL-Eye-Driver-5.3.0.0341.

mmosoll, 2025-01-01:
Camera: Elgato Facecam Pro
Windows 11

1. camera-based trackers:
Neuralnet: all is OK
PointTracker: KO => no image
Aruco: KO: KO => no image

2. Others
Eyeware Beam: all is OK
Tobii Eye tracker: all seems OK  (but usually I don't use it with OpenTrack)

renzhefeilong, 2025-01-08:
The libusbK driver of PS3 Eye camera cannot be installed, PS3 Eye open driver cannot be used, it can be replaced with the old PS3 Eye open driver, and the other cameras are used normally, and no problem has been found for the time being.

jbswsh, 2025-01-22:
Running the pre-release and everything seems fine. Both Neuralnet@60fps and Eyeware Beam@120fps run with no issues. All neuralnet settings react fine and dandy on my side.

mjcarter, 2025-04-07:
2024.1.1 worked fine for me after first installing but then after restarting PC the PS3eye disappeared from the Pointtracker settings window, like only OBS Virtual Camera appears, multiple restarts and driver reinstallations and swapping the subprocess and no luck.

Reinstalled 2023.3 and that worked fine straight away