opentrack/opentrack #1978: Assetto Corsa EVO

Kobra-361, 2025-01-17:
So in ac evo you got the option for trackIR so i thought i could use opentrack in combination with smooth track for head tracking but opentrack always says unknown game

luizgfpf, 2025-01-22:
I’m facing the same issue. I’m using OpenTrack and BeamEye, but I’m unable to get them to work with Assetto Corsa Evo. Can you please provide some guidance on how to resolve this problem?

jbswsh, 2025-01-22:
EVO does not currently work with regular trackIR devices, the current game patch has the option included but not the actual code required to run. I would keep an eye on the forums in Steam/Reddit regarding this, most up to date information is on the games discord server if you really want to be on the cutting edge.

https://steamcommunity.com/app/3058630/discussions/0/756141976595767383/
https://www.reddit.com/r/assettocorsa/comments/1i2v0r5/evo_and_track_ir/
https://discord.gg/2HXEqhM5A8

Kobra-361, 2025-01-23:
@jbswsh ok, thanks for the reply

closed 2025-01-23 by Kobra-361