opentrack/opentrack #1980: neuralnet tracker - support for higher framerates

jbswsh, 2025-01-22:
Hello,
Currently neuralnet tracker only accepts a 60 fps input/processing rate, with options such as Logitech Brio being available for almost a decade now and being able to send a 720p 90fps or 640x480 120 fps signal it would be beneficial to be able to use a higher fps option to provide smoother input when using that headtracking method. I would kindly ask for your consideration in implementing higher fps options in the software as this would benefit those who have a high fps capable webcam.

Thanks,
J

mmosoll, 2025-05-18:
I don't want discuss if useful or not (as not possible to test) but to compare with another tracker, the Eyeware Beam (with Elgato Facecam) allows only 60fps but resolutions until 4k.

In all case, honestly, I think high resolutions or higher fps>60 can unnecessarily increase computational demands without a real benefit on the detection or enhance the fluidity (taking into consideration also the framerate stability of the camera/driver at high fps)

sthalik, 2025-05-18:
Gonna add 90 and 120. What other resolutions and framerates should be included?

jbswsh, 2025-05-18:
@sthalik I'm not sure, but those have become pretty common. I am not a working developer at this moment and my knowledge is a bit out of date but software like OBS ([Open](https://github.com/obsproject/obs-studio)) is able to pull the list of resolutions and framerates straight from the driver/device information table. I am unsure if such an implementation is viable for neuralnet but that would be my suggestion going forward to future proof the software package.

**To directly answer your question** those 120 and 90 fps options seem to have become more common and would be a reasonable addition. While the list of cameras is not that long it would most definitely benefit everyone with face/eye tracking usage in mind. One special camera I know of that might benefit from a different value is the PS3 Eye camera that can go up to 150 fps @320x240 although I think it would be a very niche value to have.

And thank you for your time and consideration.

@mmosoll I won't disagree at all with your statement. It does increase demands on the hardware side but there are plenty of titles that leave headroom on the CPU which could be used for a smoother/more responsive camera feel. It is also a fair statement that CPU power will increase over time and even more headroom will be available. I have Eyeware Beam as well and have been able to use my Logitech Brio @120fps 640x320, while there may be benefits to running the camera at a higher resolution I have found the increased framerate (and in turn reduced response time/smoothness of motion) preferable. I will also state that the default should stay at 30 or 60 fps for most users, this is more of a request as a power user who likes to tinker with the available parameters.

[image]