opentrack/opentrack #1981: 60 FPS  max on PS3 eye cam?

kamransh, 2025-01-25:
Just installed PS3 eye cam however, I'm only seeing an option to select max 60 fps in Camera Configuration. How to I select more?

[image]

Len62, 2025-01-25:
What version of OT are you using? I noticed the options window layout on mine is different. I can get 75 FPS on my set-up.

kamransh, 2025-01-25:
I’m on 2023.3.0 and and followed this  https://github.com/opentrack/opentrack/wiki/PS3-Eye-open-driver-instructions

I don’t see the screen in step 5

kamransh, 2025-01-26:
> What version of OT are you using? I noticed the options window layout on mine is different. I can get 75 FPS on my set-up.

Are you using neuralnet? I think neuralnet only allows up to 60 fps which is a shame

Len62, 2025-01-26:
No, I use point tracker. I guess that is why the options window is different.