opentrack/opentrack #1986: Ps3 camera - Elite dangerous - screen shakes

nez52, 2025-01-30:
Hello,

 I’m trying to set this up for the first time but I find the screen is constantly shaking and not stable. Is this a known issue or does anyone know how I can fix it.

Thanks

Len62, 2025-01-31:
Take a look at the mapping settings. You may need to establish some dead zone space for the axis. 

[image]

willhuy, 2025-12-18:
You might also want to play with the exposure setting of the PS3 Eye camera, the shake/jittering could possibly coming from the noise lighting the camera is picking up. 

I had that issue before, and I decrease the exposure of the camera lens to the point where the only bright lighting is the 3 infrared points of the clip, you could do this in the PS3 Eye Camera Driver application. Plus play with the pixel density size recognition setting in Opentrack.