opentrack/opentrack #1990: Onnxruntime >1.19.2 breaks Neuralnet on Linux

JT8D-17, 2025-02-21:
Not sure if this is Arch Linux specific, but if I update the onnxruntime package to a release newer than 1.19.2, the neuralnet tracker fails to start. I've tried rebuilding opentrack against ONNX 1.20 and newer multiple time without success. The only thing that helps is downgrading to 1.19.2.

Can any other Linux users confirm this?

sthalik, 2025-02-21:
What happens when you try to build against 1.20?

JT8D-17, 2025-02-21:
As I've said: Neuralnet is not found.

Which ONNX version does the Windows release use?

sthalik, 2025-02-21:
It's the kind of thing that only @DaWelter knows how to fix. He should come here, eventually.

DaWelter, 2025-02-27:
:-D No idea tbh. I'll try 1.20 with my Ubuntu setup.

ThatMightyGuy, 2025-02-27:
I don't think that's helpful, but I'm having the exact same issue, down to the letter and fix. Downgrading to `onnxruntime-rocm 1.19.2` and clean rebuilding did work.

DaWelter, 2025-03-01:
Worked for me.
```
ldd ./libexec/opentrack/opentrack-tracker-neuralnet.so 
        linux-vdso.so.1 (0x00007fff42bb8000)
        libonnxruntime.so.1 => /opt/onnxruntime-linux-x64-1.20.1/lib/libonnxruntime.so.1 (0x00007f1445933000)
```
```
└─▶ ./bin/opentrack 
DEBUG [/folders/opentrack/tracker-neuralnet/ftnoir_tracker_neuralnet.cpp:494]: Loading pose net  "/tmp/opentrack/libexec/opentrack/models/head-pose-0.3-big-quantized.onnx"
DEBUG [/folders/tracker-neuralnet/model_adapters.cpp:236]: Pose model inputs ( 1 )
DEBUG [/folders/tracker-neuralnet/model_adapters.cpp:237]: Pose model outputs ( 6 ):
DEBUG [/folders/opentrack/tracker-neuralnet/model_adapters.cpp:247]:          pos_size  ( std::vector(1, 3) ) dtype:  1   ok
```
Are you also using the runtime from https://github.com/microsoft/onnxruntime/releases/tag/v1.20.1?
If not could you try it?
Just unpack and point cmake to it.

shreven0, 2025-03-01:
This is likely an issue with the Arch Linux package (see https://gitlab.archlinux.org/archlinux/packaging/packages/onnxruntime/-/issues/3). I was having the same problem, and downloading onnxruntime from https://github.com/microsoft/onnxruntime/releases/tag/v1.20.1 as @DaWelter said fixed it for me

ThatMightyGuy, 2025-03-01:
Thanks for your support. Going to try the Microsoft release later. There is a [bug report](https://gitlab.archlinux.org/archlinux/packaging/packages/onnxruntime/-/issues/3) on Arch's onnxruntime package, gonna have to wait for the fix to roll out.

ThatMightyGuy, 2025-03-12:
I've followed [this comment](https://aur.archlinux.org/packages/opentrack#comment-958865) and it worked flawlessly.

cprn, 2025-03-13:
> Did you have any luck resolving this? I tried downgrading to 1.19.2 but still had the issue, I thought I pointed to it correctly with cmake but maybe not?
> 
> edit: Apperently its still trying to point to the onnxruntime install in usr/lib even after specifying otherwise in the cmake build process, any idea how i can get it to point to the 1.19.2 version i unpacked elsewhere?
> 
> [image]

Downgrading must work, because on Manjaro (Arch based) I circumvented the issue by downgrading to `1.19.2` and reinstalling `opentrack-git` from AUR.

JT8D-17, 2025-03-16:
Still stuck on 1.19.2.  Since onnxruntime's dependency chain is apparently being updated, next to `onnxruntime` (1.19.2-4), the `abseil-cpp` package must be downgraded to 20240722.1-1 now. This still throws a strdout error, but tracking at least works.

With `onnxruntime` 1.20.2-4, the situation remains unchanged from https://github.com/opentrack/opentrack/issues/1990#issuecomment-2675435479 .

Adding the onnx paths manually to cmake as per [thecomment on the AUR package](https://aur.archlinux.org/packages/opentrack#comment-958865) does not help either.

cprn, 2025-03-16:
It's the same on Manjaro. Following the [official compilation instructions](https://github.com/opentrack/opentrack/wiki/Building-on-Linux#building-on-manjaro) doesn't work. It seems like I could make [opentrack-git](https://aur.archlinux.org/packages/opentrack-git) AUR package work if someone told me which change needs to be reverted between `onnxruntime` versions `1.19.2` and `1.20.2` - I think it's one `if` somewhere that caused the issue (I'm guessing it's some kind of check to see if `onnxruntime` should link to `onnx` statically or dynamically, but I failed to find it).

JT8D-17, 2025-03-25:
With `onnxruntime` 1.20.2-7 from Arch's repositories (and after recompiling opentrack against it), neuralnet works again for me. Can anybody confirm?

cprn, 2025-04-08:
Yes, it works fine compiled against 1.20.2-7, the issue can be closed IMHO.

closed 2025-04-08 by JT8D-17