opentrack/opentrack #2009: crashes when compiled with LTO

kakra, 2025-04-06:
On Gentoo, I tried to compile opentrack with the following flags:
```bash
WARNING_FLAGS="-Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing"
COMMON_FLAGS="-O2 -march=native -fomit-frame-pointer -g -ftree-slp-vectorize -flto=auto ${WARNING_FLAGS}"

CFLAGS="${COMMON_FLAGS}"
CXXFLAGS="${COMMON_FLAGS} -fvisibility-inlines-hidden"
LDFLAGS="-Wl,-O1 -Wl,--as-needed -Wl,--sort-common -Wl,-z,pack-relative-relocs ${WARNING_FLAGS}"
```

Compiling worked, but starting opentrack results in each widget having a red border. And clicking most buttons results in a segfault.

Compiling without LTO fixes the issues:
```
-- <<< Gentoo configuration >>>
Build type      RELWITHDEBINFO
Install path    /opt/opentrack
Compiler flags:
C               -O2 -march=native -fomit-frame-pointer -g -ftree-slp-vectorize -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing
C++             -O2 -march=native -fomit-frame-pointer -g -ftree-slp-vectorize -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -fvisibility-inlines-hidden -fopenmp
Linker flags:
Executable      -Wl,-O1 -Wl,--as-needed -Wl,--sort-common -Wl,-z,pack-relative-relocs -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing
Module          -Wl,-O1 -Wl,--as-needed -Wl,--sort-common -Wl,-z,pack-relative-relocs -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing
Shared          -Wl,-O1 -Wl,--as-needed -Wl,--sort-common -Wl,-z,pack-relative-relocs -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing
```

sthalik, 2025-04-08:
Sounds like an obvious case of miscompilation. But try passing `-flto-partition=none`, it used to help for me.

kakra, 2025-06-29:
@sthalik I retried with the Qt6 version and it seems fine now: It compiles and runs without crashes when LTO is fully enabled. Thanks. Feel free to close then.

closed 2025-06-29 by sthalik