opentrack/opentrack #2013: "Unknown game" even without any game open

Maronho, 2025-04-12:
Hello,

I'm encountering an issue where OpenTrack displays "Unknown game" even when no game is running. I've tried the following solutions without success:

    Fresh Windows 11 installation: I just did a fresh install of Windows 11, no old profiles or config files

    Checked output protocol settings: 

I've ensured that the output protocol is set correctly (e.g., "Freetrack 2.0 Enhanced" for DCS World), and I've tried switching between different output protocols, but the issue persists.

    Checked administrator permissions: Both OpenTrack and the game are set to run with the same permissions (either both as administrator or both as regular user).

    Reinstalled OpenTrack: I've uninstalled and reinstalled OpenTrack, but the problem persists.

I would appreciate any help or suggestions to fix this issue. My system is running Windows 11, and I have already installed all necessary Visual C++ Redistributable libraries.

Thank you in advance!

Len62, 2025-04-12:
If there is NO game running it will display "Unknown Game".

Maronho, 2025-04-12:
Thanks in advance.

I misspoke, sorry. I’ve tried it in at least 2 games that, under other circumstances, have been natively compatible with freetrack: Microsoft Flight Simulator 2024 and Assetto Corsa Competizione.

When I open them, the game doesn’t recognize the presence of a headtracking device, and the OpenTrack window shows "unknown game" without changing to the game’s name.

When I was on Windows 10, this didn’t happen.

Thanks again, and let’s see if we can come to a solution.

sthalik, 2025-04-12:
The database hasn't been updated in a long while.

If you want to help, download the TrackIR software, update its game list and send the file `sgl.dat` from your user's AppData (not Program Files).

Len62, 2025-04-12:
Not sure if this is what you want to do. On the Options, Game Detection tab you can directly link the games .exe file. With OT running, you then start the game and OT should start tracking.

[image]