opentrack/opentrack #2018: Increasing the EWMA filter values

hhelikopter, 2025-04-22:
I need help. I use Opentrack in the MSFS2024 simulator. I use the EWMA filter which is very pleasant for me.

I would like to further increase the minimum, maximum, curve values ​​of the filter in order to achieve slower movement, in order to make the movement even slower and smoother.

It would be good if the minimum, maximum value sliders were increased to 200%, and the curve value slider were increased to 1000%.

Thank you very much

kakra, 2025-04-22:
You could try this profile which I am using for Elite Dangerous:
https://gist.github.com/kakra/02a2679919f1ce5c0e4e92f49b06f60e

It requires the latest pre-release due to the Accelera Hamilton filter. It uses very smooth movement, no dead area in the center for more natural operation, and a wide angle. Does this help you?

What's the filter you are using? Which settings?

hhelikopter, 2025-05-05:
Thank you for your answer.