opentrack/opentrack #2023: Games with tight NPClient.dll requirements

kennethfm, 2025-04-30:
Hi @sthalik !

I'm trying to understand how OpenTrack functions when it comes to the TrackIR protocol, so I have a couple of questions, if you don't mind:

- Have been able to established which games actually use the contents from `NP_GetSignature`, and which games ignore it? A priori, it looks like checking for `NP_GetSignature` which looks like to be a point of contention, is merely optional from the game developer point o view.
- How were you able to obtain the table codes per game listed in `facetracknoir supported games.csv` ?

Thank you very much for the opentrack project, and for your answers!

sthalik, 2025-04-30:
The table can be extracted using the [game_data](https://github.com/opentrack/opentrack/tree/master/contrib-noinst/important-stuff) code. Once in the text format, you pass it together with the existing .csv to [make-csv.pl](https://github.com/opentrack/opentrack/blob/master/contrib-noinst/make-csv.pl), which makes a merged .csv. 

Most of the work was done by the amazing @uglyDwarf.

For `NP_GetSignature`, the story was that some of the bigger devs figured out the anti-competitive maneuver that NaturalPoint was doing, then disabled that. I don't presume a lot of them did. If you really want to know you can build your own `NPClient.dll`.

kennethfm, 2025-04-30:
Thanks for the answer! 

Awesome. But then, what is the input file to `game_data.exe`? I can’t seem to find that. Also I see that the latest update to `facetracknoir supported games.csv` was 4 years ago. Is the protocol changing?

Perhaps I’ve missed a table out there of supported games per protocol, but I haven’t had success getting games running with opentrack when the trackir protocol is disabled (only freetrack 2.0 is enabled). So, which games are compatible in this configuration? Do such games contain the “official “ NPClient.dll and the freetrack protocol is actually compatible with it by feeding data to it, or is it assumed that the games actually link against the freetrack client DLL? I guess I am trying to understand if the freetrack protocol, as implemented by opentrack, actually depends on some version of NPClient.dll to exist or if it can be considered completely independent. And if it is completely independent, then why is the game data/csv/table required in the freetrack implementation…

Thank you kindly for helping with my many questions

sthalik, 2025-04-30:
> what is the input file to game_data.exe

`sgl.dat` from TrackIR software. When run from the Windows console you can redirect it to a file: `game_data sgl.dat > sgl.txt`.

> haven’t had success getting games running with opentrack when the trackir protocol is disabled (only freetrack 2.0 is enabled). So, which games are compatible in this configuration? 

There was a non-zero amount of games using the FreeTrack protocol.

> I guess I am trying to understand if the freetrack protocol, as implemented by opentrack, actually depends on some version of NPClient.dll to exist or if it can be considered completely independent.

`npclient.c` is in the tree.