opentrack/opentrack #2030: opentrack 2025.1 release process

sthalik, 2025-05-17:
Hey,

Last year there wasn't a single fully-functional release. Hopefully not this year.

Here are some topics of interest:

- update to Qt 6 **MOSTLY DONE**
- go through bug reports and feature requests and implement at least some of them **NOT DONE**

And things that likely aren't code issues but build and dependency problems:
 
- verify fixing camera not opening bugs (both `video/opencv` and `video/ps3eye`)  **NOT DONE**
- verify no more issues with loading modules **NOT DONE**

I think it's a combination of:

- the disastrous attempt of switching the MSVC++ build from `-MT` to `-MD`.
- using Qt 5 fork from KDE (at some point it stopped working well on Windows)
- Qt 5 becoming unmaintained and obsolete

There are definitely legitimate issues with e.g. opening the cameras but hopefully we'll be able to get rid of issues that only started existing in the 2024 build.

Then we can think about the major future architectural changes (unit tests, adding *all* dependencies as submodules, including Qt and ONNXRuntime).

sthalik, 2025-05-18:
Here's the first build of the year. Please report if there are any regressions since 2023, and whether cameras (DirectShow, libusb) open properly.

https://aurora.misaki.pl/opentrack/opentrack-test-20250517.7z

mmosoll, 2025-05-18:
First short testing in Windows11 with Eyeware Beam and Neuralnet tracker seems correct => I will update the comment as soon as tested more deeply.

Note: Title bar still indicating 2024.1.1 but it is not an issue...
Thanks for this new version

sthalik, 2025-05-18:
If someone can test the PS3 Eye open driver, please drop a note. There shouldn't be any need for workarounds such as with `ps3eye-subprocess.exe`.

Note that the CL-Eye driver doesn't work because it's a 64-bit build. I'll see how many complaints there are and in the worst case scenario add a subprocess-based camera for 32-bit-only DShow filters.

sthalik, 2025-05-19:
I'll be merging https://github.com/opentrack/opentrack/tree/feature/qt6 in a week or so unless someone has good reason not to.

Berzerker3009, 2025-05-20:
Hey , sthalik love the software, wanted to ask about commissions do you take commissions to implement features to it , for a new output method to control controller analog camera movement through ds4 windows

renzhefeilong, 2025-05-23:
> Here's the first build of the year. Please report if there are any regressions since 2023, and whether cameras (DirectShow, libusb) open properly.
> 
> https://aurora.misaki.pl/opentrack/opentrack-test-20250517.7z

It cannot be downloaded under the Chinese network.

sthalik, 2025-05-24:
Actually rather than a third-party site try this instead: https://ananke.misaki.pl/opentrack/opentrack-test-20250517.7z

renzhefeilong, 2025-05-24:
> Actually rather than a third-party site try this instead: https://ananke.misaki.pl/opentrack/opentrack-test-20250517.7z

Thank you very much, downloaded.

renzhefeilong, 2025-05-24:
opentrack-test-20250517 Test Report

1. Under the Windows 10 system, the "neuralnet tracker" and "TrackHat v1 Sensors" options are missing in the input settings. On Windows 11, "neuralnet tracker" is present, but "TrackHat v1 Sensors" remains unavailable.
2. The interface fonts are less clear compared to previous versions.
3. PS3 Eye Open Driver function properly when libusb or CL-Eye-Driver is installed.
4. The head-pose-0.3-big-quantized.onnx model has a lower frame rate and slightly higher CPU usage/inference time than head-pose-0.2-big.onnx. It is recommended to default to head-pose-0.2-small.onnx for compatibility with lower-specification computers.
Suggested defaults for Head Center Offset: Forward: -30mm, Up: -40mm, Right: 0mm. Enabling MJPEG by default is advised, as many cameras achieve maximum frame rate only in MJPEG mode.
5. The expanded frame rate options for neuralnet tracker are a notable improvement.

kakra, 2025-06-29:
The Gentoo build from the current Git snapshots works flawlessly with webcam inference, 30fps (maximum of my webcam according to v4l although it should support 60fps at that low resolution), 2-3ms inference latency (head post 0.2 small). Proton integration works, too. It's stable.

kixeou, 2025-07-06:
cant open

saimat, 2025-07-26:
Is there any progress or new builds since May that we can test?

sthalik, 2025-07-26:
There's [a build](https://ananke.misaki.pl/opentrack/opentrack-test-20250726_01.7z). It implements gamepad key shortcuts. Of course, keyboard/mouse/joystick shortcuts should function the same.

renzhefeilong, 2025-07-27:
> There's [a build](https://ananke.misaki.pl/opentrack/opentrack-test-20250726_01.7z). It implements gamepad key shortcuts. Of course, keyboard/mouse/joystick shortcuts should function the same.

The link won't open.

sthalik, 2025-07-27:
@renzhefeilong sorry. Now it opens.

renzhefeilong, 2025-07-27:
> [@renzhefeilong](https://github.com/renzhefeilong) sorry. Now it opens.

Old question:
Windows 10 cannot use the neuralnet tracker.
New question:
The shortcut cannot be bound, and the mouse button is automatically bound after pressing the left mouse button.

sthalik, 2025-07-28:
Maybe today. https://ananke.misaki.pl/opentrack/opentrack-test-20250728_01.7z

Neuralnet should work normally on Windows 10. In fact I'm using it right now. Can you describe your software and hardware configuration?

sthalik, 2025-07-28:
Use today's build as it had onnxruntime updated to the latest stable version.

Replace `onnxruntime.dll` from their github [site](https://github.com/microsoft/onnxruntime/releases/tag/v1.22.1.).

renzhefeilong, 2025-07-28:
[image]

sthalik, 2025-07-28:
@renzhefeilong please try this special build: https://ananke.misaki.pl/opentrack/opentrack-test-20250728_01-onnx-only.7z

sthalik, 2025-07-28:
@renzhefeilong please see if the following build works: https://ananke.misaki.pl/opentrack/opentrack-test-20250728_03.7z. I'd prefer using static CRT and sadly there are no such official builds.

sthalik, 2025-07-29:
I'm trying something new for numpad keys. They should work but also combinations with Ctrl/Alt/Shift should now be recognized.

@renzhefeilong, can you confirm if the yesterday's or today's build works with neuralnet for you? It influences whether I can use the static CRT.

https://ananke.misaki.pl/opentrack/opentrack-test-20250729_01.7z

renzhefeilong, 2025-07-29:
@sthalik 20250728_03 version of neuralnet is available and has a lower CPU usage, especially the 0.3 model, but I don't know if the accuracy will be reduced or not.

renzhefeilong, 2025-07-29:
@sthalik ,
https://ananke.misaki.pl/opentrack/opentrack-test-20250729_01.7z Used normally, no problems have been found for the time being.

sthalik, 2025-07-30:
If anyone has problems with shortcuts (gamepad/keyboard/mouse/joystick), please tell me how to reproduce them.

sthalik, 2025-07-30:
https://github.com/opentrack/opentrack/issues/2030#issuecomment-3130226559 Looks like something doesn't get properly converted from Unicode to QString.

sthalik, 2025-07-31:
Following up from how `numeric.hpp` now uses doubles, how about a quaternion implementation that uses doubles as well?

renzhefeilong, 2025-07-31:
> If anyone has problems with shortcuts (gamepad/keyboard/mouse/joystick), please tell me how to reproduce them.

I didn't identify any new shortcut-related issues, but there's a persistent problem worth noting: When using OpenTrack, if you press the Ctrl+Alt+Del combination, the software erroneously detects both Ctrl and Alt keys as being held down. This detection glitch causes all subsequently assigned shortcuts to automatically combine with Ctrl+Alt modifiers, overriding the original shortcut configurations. The system returns to normal functionality after manually pressing both Ctrl and Alt keys separately to reset their input states.

Langolyers, 2025-08-03:
> If anyone has problems with shortcuts (gamepad/keyboard/mouse/joystick), please tell me how to reproduce them.

Not sure if related and apologies if I'm hijacking the discussion, but shortcuts are not working for me on Linux. 

When I assign them they are recognized and update, but when using them in-game they do not work at all.

sthalik, 2025-08-30:
@quanbingyi,

Unfortunately I can't reproduce it on my PC. Because of that I can't directly fix it per your report. What I can do is give you a list of questions and later some test builds to try and fix it.

1. Does it fail when trying to bind the key, or when trying to actually use it after binding it?
2. If you're able to bind it, does the text say Num+5, like so? <br>
    [image]

I'm making a testing build for you but let's try the more basic possibilities first.

quanbingyi, 2025-08-30:
I'm extremely eager to get the beta version.

sthalik, 2025-08-30:
Try this:

1. Get the build: https://ananke.misaki.pl/opentrack/opentrack-test-20250830_01.7z
1. Select Profile -> 'Create new empty' config on the main window.
1. Exit opentrack.
1. Open a Windows console window, for example by pressing Win+R, typing in `cmd` and pressing Return.
1. Drag `opentrack.exe` onto the console window and press Return.
1. From now on it's best to navigate everything using only the mouse, unless expressly needed.
1. Now on this new empty profile you've created, go to Options and set the keybinding for 'Restart tracker'. There shouldn't be any more keybindings than the one you've just created.
1. Select the "Test tracker" (input tracker) on the main screen.
1. Press Numpad 5, once with num lock and another without.
1. Go to the console window, then press Alt+PrintScreen to get the entire content. For example: <br>
    [image]
1. Optional: Verify that the screenshot is OK in an image program.
1. Ctrl+V the screenshot you have into this GitHub input box, then send it.

sthalik, 2025-08-30:
@quanbingyi, is this a laptop with the Fn Lock enabled?

renzhefeilong, 2025-08-30:
> 20250729_01 hotkey Number5 is not work, The other is fine.

我这边没有遇到你说的这个问题。

quanbingyi, 2025-08-30:
我也纳闷，看起来别人都正常，就我的不正常，2024一切ok，2025.所有版本，回中不响应，2025.7.29那个重新追踪就会挂掉

quanbingyi, 2025-08-30:
i don't know ,but it working  very well, maybe  system is update  ,

renzhefeilong, 2025-08-30:
> It hasn't been committed yet—it's only halfway implemented. I'm going to rephrase it in English before sending a test build.
> 
> The feature is for mounting the camera asymmetrically. If you have the camera mounted on top of the display, moving closer to the screen on the Z axis will shift your Y axis. If you have the camera mounted to the side of the screen, moving toward the screen will rotate and move you on the horizontal axis.
> 
> This feature is meant to work mostly for position changes, but with some more implementing and testing it might turn out to be also useful for affecting rotation

To resolve this issue, displacement calculations should use the raw head orientation as the reference basis instead of relying on the calibrated "Little Octopus" reference. Pressing the "Center" button should only reset all axis data without modifying orientation calibration. This ensures no recalibration is needed regardless of camera placement.

sthalik, 2025-09-03:
Centering indeed uses raw data. Mapping happens a bit later. 

I'm going to look into finishing the work on the camera offset, hopefully soon.

renzhefeilong, 2025-09-03:
Positioning should be based on the head itself, rather than relying on camera-captured images. For example:

If cameras are placed on both sides, the captured head position appears tilted in the footage.
With head-based positioning: All axes will rotate in sync with the head tilt, maintaining system integrity (no adverse effects).
With camera-image positioning: While all axes remain "aligned" in the system, a tilted head in reality creates significant alignment conflicts.
Additional critical distinctions:

Head-based positioning:

Requires no calibration (works intrinsically).
Cameras can be freely offset (position-adjusted) without system impact.
Camera-image positioning:

Cameras cannot be offset (position adjustments directly distort perceived head orientation).
Any camera displacement would severely disrupt system functionality.

renzhefeilong, 2025-09-29:
> I'm trying something new for numpad keys. They should work but also combinations with Ctrl/Alt/Shift should now be recognized.
>
> [@renzhefeilong](https://github.com/renzhefeilong), can you confirm if [...]

After two months of use, it has been working well. Personally, I believe it's ready for an official release.

renzhefeilong, 2025-10-01:
> > I still need to test the camera offset option. It likely won't work at the first time. It's kind of difficult to get anything done for me though.
> > Feel like providing feedback? See [47c5722](https://github.com/opentrack/opentrack/commit/47c5722dd02ac83309e424cb7bd6a0505ec9e48a) or the binary [build](https://ananke.misaki.pl/opentrack/opentrack-test-20250926_01.7z). It might be that it only works properly for position data and for rotation it needs to be disabled.
> 
> Tried it out, it works to some extent but doesn’t seem very accurate. Probably due to incorrect angle settings.

After configuring the correct parameters, the results are acceptable. However, the setup is quite complicated, making it difficult to determine the optimal settings. It would be ideal if pressing the center key could calibrate the offset angle without having to manually adjust it.

sthalik, 2025-11-06:
You can now apply the camera offset more easily as from now on it only applies to position tracking. The code for rotating the rotation was incorrect

See if this build works [opentrack-test-20251106_01.7z](https://ananke.misaki.pl/opentrack/opentrack-test-20251106_01.7z). If it does then I'm going to releaase it  with no changes. Well, except for giving @renzhefeilong time to update the Chinese translation.

quanbingyi, 2025-11-16:
20251106
DiRT Rally 2 1.18   crash
city car driving 1.59  crash
the others, sometimes novideo,

CSV Data logging Enable ,will crashed sometimes

renzhefeilong, 2025-11-17:
> I use a clip mounted on the left side of my headset. Not sure which angle size you are referring to. I agree that the camera offset may not help when using pointtracker. With the camera offset turned off I have no trouble with tracking. I may try experimenting some more just to see.

@Len62 
What are the Yaw, Pitch, and Roll values? I'd like to try your parameters. Here's my test result from 20250926 - there are minor deviations which I believe stem from the body's inherent physical offsets.

[image]
[image]
[image]

sthalik, 2025-11-30:
In this build you can rotate not just the position vector but also the rotation itself. See if this works according to your needs. https://ananke.misaki.pl/opentrack/opentrack-test-20251130_01.7z

The b5d9e9a2 build looks good.

renzhefeilong, 2025-12-02:
> Still no luck using camera offset with this build. With camera offset off tracking is perfect. As [@renzhefeilong](https://github.com/renzhefeilong) stated before it may not be of use with the point tracker.

@Len62 
My point tracker is working fine. Here is how I set it up.

[image]

Len62, 2025-12-02:
Thank you @renzhefeilong . I'll give it another try. I normally use the accela filter but maybe I'll try using the Hamilton filter. I'll report back later.

sthalik, 2025-12-12:
Here's the release candidate: https://ananke.misaki.pl/opentrack/opentrack-2025.1.0-rc1.7z. The title bar says otherwise because the Git tag hasn't been created yet.

sthalik, 2025-12-19:
This is the [2025.1.0-rc2](https://ananke.misaki.pl/opentrack/opentrack-2025.1.0-rc2.7z) build. Hopefully no more Qt issues.

renzhefeilong, 2025-12-19:
@sthalik In my mind, https://ananke.misaki.pl/opentrack/opentrack-test-20251130_01.7z is the most perfect version so far, and I have already been using it as the official version.

kakra, 2025-12-23:
For some reason, opentrack is tracking a lot worse than before, it's stuttering, and has a lot of jitter problems when I place my hands near my face. But I'm not sure if this is due to some update to opentrack (there have been changes to the pose models, I think), or if this is due to my kernel update from 6.12 to 6.18.

I think the jitter can be explained by the changed models, the stutters are more likely coming from changed scheduling behavior of the kernel maybe? Or movement between frames is just less smooth in newer version? Could it become better if opentrack would increase its own priority?

sthalik, 2025-12-23:
@renzhefeilong in [this build](https://ananke.misaki.pl/opentrack/opentrack-test-20251223_02-nolto.7z) I was able to fix almost everything. Notably, video/ps3eye doesn't work.

DaWelter, 2025-12-24:
> I'm using neuralnet on Linux. I'll try the big pose model. There are not so many left since some update: big and small, in two variants:

Could you try the old models?

Jitter when hand near face (i.e. hand occludes part of the face) sounds like normal behavior. 

But the new ones works just fine for me. From my experience, behavior when face is occluded is similar as before.

> Setting "alpha smoothing" to 0 helper with the jitter. Tho, I'm not sure why "0" means full smoothing, and "1" jitters like hell - I would expect higher value = more smoothing. It's most likely an issue with the light here but it worked on older version, so it looks like the new versions needs more light?

> "alpha smoothing" at 0.03 eliminates almost all jitter, 0.04 already introduces jitter. Only a value of 0.00 seems to be safe. "internal noise filter" seems to do nothing.

This parameter is the mixing factor for exponential weighted averaging of the facial bounding box. With a setting so low the crop area wouldn't follow your head translation well. It should work close to 1.

@sthalik I can't do any development until after new year. If there is need, you'll have to make changes. 🎄 🎁

mmosoll, 2025-12-24:
> [@mmosoll](https://github.com/mmosoll) sorry, forgot to link to the [actual build](https://ananke.misaki.pl/opentrack/opentrack-test-20251223_01-clang-asan.7z)!
> 
> Open a cmd.exe / Windows console window and drag opentrack.exe there. then execute it with Return and once done, copy or screenshot the output.
> 
> If there's a crash, it's going to emit a long, descriptive message.

Thanks @sthalik for your build and sorry for the delay in answering you.

1. Error => library "libzstd.dll" missing, but solved downloading it from github

`
opentrack.exe - System Error: The code execution cannot proceed because libzstd.dll was not found. Reinstalling the program may fix this problem. 
`

2. No PointTracker or Aruco available but OpenTracker interface is opening

`
DEBUG [D:/dev/opentrack/api/plugin-support.hpp:108]: library "aruco" failed: "Cannot load library C:\\progs\\opentrack\\modules\\\\opentrack-tracker-aruco.dll: The specified module could not be found."
DEBUG [D:/dev/opentrack/api/plugin-support.hpp:108]: library "easy" failed: "Cannot load library C:\\progs\\opentrack\\modules\\\\opentrack-tracker-easy.dll: The specified module could not be found."
DEBUG [D:/dev/opentrack/api/plugin-support.hpp:108]: library "pt" failed: "Cannot load library C:\\progs\\opentrack\\modules\\\\opentrack-tracker-pt.dll: The specified module could not be found."
DEBUG [D:/dev/opentrack/api/plugin-support.hpp:108]: library "wii" failed: "Cannot load library C:\\progs\\opentrack\\modules\\\\opentrack-tracker-wii.dll: The specified module could not be found."
DEBUG [D:/dev/opentrack/api/plugin-support.hpp:108]: library "opencv" failed: "Cannot load library C:\\progs\\opentrack\\modules\\\\opentrack-video-opencv.dll: The specified module could not be found."
`

3. Cannot start tracking as consequence of the issue 2 but when closing OpenTracker the same error message regarding overrun is displayed. No error trace indicated in the console.

`
_q_titlebar: opentrack.exe - System Error: The system detected an overrun of a stack-based buffer in this application. This overrun could potentially allow a malicious user to gain control of this application. 
`

renzhefeilong, 2025-12-24:
> Version 20251130_01 also presents an issue: after installing the Visual C++ V14 Redistributable package (version 14.50.35719), the application crashes when the NeuralNet Tracker is used.

C:\Users\Administrator>
DEBUG [D:\dev\opentrack\api/plugin-support.hpp:108]: library "kinect-face" failed: "Cannot load library D:\\Documents\\Desktop\\install\\modules\\\\opentrack-tracker-kinect-face.dll: DEBUG [D:\dev\opentrack\input\gameinput.cpp:97]: gi: load failed: "Cannot load library GameInput.dll: DEBUG [D:\dev\opentrack\input\gameinput.cpp:344]: gameinput: not available
[ INFO:0@7.187] global registry_parallel.impl.hpp:96 cv::parallel::ParallelBackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(0, sorted by priority): N/A
DEBUG [D:\dev\opentrack\tracker-neuralnet\env.cpp:57]: tracker/nn: loading onnxruntime library"D:/Documents/Desktop/install/modules/onnxruntime-avx.dll"
DEBUG [D:\dev\opentrack\tracker-neuralnet\env.cpp:60]: tracker/nn: can't load onnxruntime library : "Cannot load library D:\\Documents\\Desktop\\install\\modules\\onnxruntime-avx.dll:

sthalik, 2025-12-24:
Okay, here are more builds. Any luck?
 
- https://ananke.misaki.pl/opentrack/opentrack-test-20251224_01-msvc2022.7z
- https://ananke.misaki.pl/opentrack/opentrack-test-20251224_02-msvc2022-qt-6.9.1.7z

Now I'm out of ideas. The only things that come to mind are switching to `-MD` (multithreaded dll) or switching to clang-cl or clang.

Is there now a better result compared to the 20251223 build?

Pieloth, 2025-12-25:
I noticed also those Neuralnet jitters on latest Windows builds, compared with 2024.1 release
I tried to fix those by adjusting the Natural Movement filter, with no luck.

What I noticed however between the recent and 2024.1 Neuralnet trackers, is that the white circle around the head was much bigger on the 2024.1 release, than it is with recent builds. 

I dont't know the reason for this circle size reduction... might be a clue?

sthalik, 2025-12-25:
@Pieloth, @kakra: let's make sure tracker/neuralnet problems go away when reverting this one single commit. Please try [this build](https://ananke.misaki.pl/opentrack/opentrack-test-20251225_01-neuralnet-temp-revert.7z) and see if the jittering is gone.

@mmosoll: please check the build at https://github.com/opentrack/opentrack/issues/2030#issuecomment-3690595466 and see how many libraries still fail to load.

sthalik, 2025-12-25:
@kakra branch wip-2026.1.0

renzhefeilong, 2025-12-25:
> Okay, here are more builds. Any luck?
>
> * https://ananke.misaki.pl/opentrack/opentrack-test-20251224_01-msvc2022.7z
> * [...]

I tested 20251224_01-msvc2022, but the issue persists—there's no camera list. The 20251224_02-msvc2022-qt-6.9.1 file cannot be opened and shows a 404 Not Found error.

[13 more replies]