opentrack/opentrack #2040: arma reforger/battleeye is blocking  dll

c0ldn3s5, 2025-06-06:
18:48:07: Starting BattlEye Service...
18:48:09: Updating...
18:48:11: Launching game...
18:48:28: Note: File blocks can be ignored if they don't cause problems with the game.
18:48:28: [INFO] Blocked loading of file: "C:\Program Files (x86)\opentrack\modules\NPClient64.dll".

this was working yesterday and then today i get this.... fucking reforger, doesnt have joystick support and now they are stealing my headtracker from me

c0ldn3s5, 2025-06-06:
they cause problems with the fucking game battleeye

c0ldn3s5, 2025-06-07:
i dont get it, reinstalled opentrack and now battleeye doesnt see the dll and tracking works... i swear mother FUCKING BOHEMIA! let me fly proper god damnit!

c0ldn3s5, 2025-06-07:
worked for a second and still getting blocked and since in using delanclip and the ps3eye camera i am fucked i feel like

borathtl, 2025-06-07:
i got the same issue as well, swear to god with every update they manage to brake something

Chalta, 2025-06-07:
BI ticket raised at https://feedback.bistudio.com/T189252

sthalik, 2025-06-13:
#1985