opentrack/opentrack #2054: NeuralNet input option missing in 2024.1.1 (Linux, bazzite/Fedora 42)

bossardnutz, 2025-07-20:
Compiled in a Fedora 41 distrobox using the following commands:

```
sudo dnf install cmake git qt6-qttools-devel qt6-qtbase-private-devel procps-ng-devel opencv-devel onnxruntime-devel qt6-qtserialport-devel
git clone https://github.com/opentrack/opentrack
cd opentrack
cmake .
make
make install
cd install/bin
./opentrack
```

This produces the following output:
```
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: From 6.5.0, xcb-cursor0 or libxcb-cursor0 is needed to load the Qt xcb platform plugin.
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vkkhrdisplay, vnc, xcb, wayland-egl, wayland.

Aborted (core dumped)
```

Outside the distrobox, I can navigate to ~/opentrack/install/bin and execute the opentrack file. Opentrack launches, but the input selector only shows the following items:

- FreePIE UDP receiver
- Fusion
- Hatire Arduino
- Linux joystick input
- S2Bot receiver
- Test tracker
- UDP over network

What do I have to do to get NeuralNet into the input list?

bossardnutz, 2025-07-21:
Thanks for looking into this!

I'm very new to this - by "set in cmake" do you mean something in CMakeLists.txt?

I deleted the QUIET flag from a couple of lines in that file and when I run `cmake .` the output now includes:

`-- Found OpenCV: /usr (found version "4.10.0")
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Found ONNXRuntime: /usr/lib64/libonnxruntime.so`

Looking through the files, I can see the following file:
`~/opentrack/tracker-neuralnet/opentrack-tracker-neuralnet.so`

miktolon, 2025-09-27:
I got neuralnet tracker working from Fedora distrobox by exporting `LD_LIBRARY_PATH`. Built with onnxruntime 1.23.0.

```
export LD_LIBRARY_PATH="$HOME/opentrack/onnxruntime-linux-x64-1.23.0/lib:$LD_LIBRARY_PATH"

ldd "$HOME/opentrack/install/libexec/opentrack/opentrack-tracker-neuralnet.so" | grep -i onnx
# libonnxruntime.so.1 => $HOME/opentrack/onnxruntime-linux-x64-1.23.0/lib/libonnxruntime.so.1 (...)
```

After that, neuralnet tracker appeared in the UI dropdown. Should be same, just check the version using.

I also built natively on Bazzite (so no need for distrobox).

Install packages (then reboot):

```
sudo rpm-ostree install \
  cmake gcc-c++ git pkgconf-pkg-config \
  qt6-qtbase qt6-qttools qt6-qtmultimedia \
  qt6-qtbase-devel qt6-qttools-devel qt6-qtmultimedia-devel \
  opencv opencv-devel eigen3-devel libproc2-devel \
  libX11-devel libxcb-devel xorg-x11-proto-devel \
  mesa-libGL-devel mesa-libEGL-devel

systemctl reboot
```

Clone and build:
  
```
git clone https://github.com/opentrack/opentrack.git "$HOME/opentrack"
cd "$HOME/opentrack"

# fetch onnxruntime to same location
curl -LO https://github.com/microsoft/onnxruntime/releases/download/v1.23.0/onnxruntime-linux-x64-1.23.0.tgz
tar -xzf onnxruntime-linux-x64-1.23.0.tgz

cmake -S . -B build-host \
  -DCMAKE_BUILD_TYPE=Release \
  -DCMAKE_INSTALL_PREFIX="$HOME/opentrack-host/install" \
  -DONNXRuntime_DIR="$HOME/opentrack/onnxruntime-linux-x64-1.23.0" \
  -DONNXRuntime_INCLUDE_DIR="$HOME/opentrack/onnxruntime-linux-x64-1.23.0/include" \
  -DONNXRuntime_LIBRARY="$HOME/opentrack/onnxruntime-linux-x64-1.23.0/lib/libonnxruntime.so" \
  -DCMAKE_INSTALL_RPATH='$ORIGIN/../lib64;$ORIGIN/../lib;$ORIGIN/../libexec/opentrack;$ORIGIN/../libexec/opentrack/onnx' \
  -DCMAKE_BUILD_RPATH='$ORIGIN'

cmake --build build-host -j
cmake --install build-host
```

Because the system may have different ONNX version, created small launcher that prepends the local 1.23.0 to `LD_LIBRARY_PATH`:

```
install -Dm755 /dev/fd/3 ~/.local/bin/opentrack-host 3<<'SH'
#!/usr/bin/env bash
set -euo pipefail
PREFIX="$HOME/opentrack-host/install"
ONNX="$HOME/opentrack/onnxruntime-linux-x64-1.23.0"
export LD_LIBRARY_PATH="$ONNX/lib:$PREFIX/lib64:$PREFIX/lib:$PREFIX/libexec/opentrack:$PREFIX/libexec/opentrack/onnx:${LD_LIBRARY_PATH-}"
export QT_QPA_PLATFORM="${QT_QPA_PLATFORM:-wayland;xcb}"
exec "$PREFIX/bin/opentrack" "$@"
SH

# run it
opentrack-host
```