opentrack/opentrack #2055: Mac and Euro Truck Simulator

rak00001, 2025-07-20:
Hey all, especially the awesome @matatata, I was wondering if there's plans on making this able to work with ETS / ATS? Just hoping that's not a thing that's impossible? :)

Pieloth, 2025-07-20:
What are your ETS2/ATS issues on Mac? 
OT works perfectly on Win11 for these 2 sims, using it for years with basic Webcam + Neural Net

matatata, 2025-07-25:
I have an update regarding the WINE-integration that I think is probably the best chance to play HeadTracking-Games on a Mac. Like I said it's very complicated given the fact that macOS is such a niche in many regards and every time I do things that worked fine before just break. And those things are needed to develop and test all that. However after many hours I found that indeed I was able to tweak things until I could use Headtracking (TrackIR protocol) in a Windows Game (Falcon BMS) run under CrossOver 24.0.7 or CrossOver 25.0.1. So since CrossOver probably is the most reliable and stable platform playing Windows Games on a Mac this really is good news. CrossOver costs money, bit I think it is well spent - if you insist on playing Games on a Mac that is of course ;-) ...

It does not work on stock wine 10.x.. only wine 9.0...

So I'll see what can be done to facilitate getting it to play nice with CrossOver and then there'll be an update.

matatata, 2025-12-24:
ETS2 run via CrossOver can now be fed by MacOpentrack using https://matatata.gumroad.com/l/opentrackwinebridge