opentrack/opentrack #2059: no gui on startup linux

silverlore, 2025-08-03:
I have just build the master branch on my Kubuntu 25.04 according to the "Building on Linux" guide. When I get to the last line ./opentrack, I do not get any Gui. 

I do not get any message in the terminal and I can see in system monitoring that opentrack is running and consuming 6 percent of cpu processing but nothing happens. I have checked the tray and it is not showing any icon there either.

I hope Someone can help me.

mgomersbach, 2025-08-09:
Same here, strace'd it, first it hung on accessing /etc/timezone, after fixing that by temporarily setting TZ env var (glibc issue not opentrack), but after that it still doesn't show a window, tries to read config, closes file and just hangs there:

```
access("/home/mark/.config/opentrack-2.3/default.ini", F_OK) = 0
openat(AT_FDCWD, "/usr/bin/../libexec/opentrack", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 19
fstat(19, {st_mode=S_IFDIR|0755, st_size=3488, ...}) = 0
getdents64(19, 0x559f5923ac30 /* 37 entries */, 32768) = 1608
getdents64(19, 0x559f5923ac30 /* 0 entries */, 32768) = 0
close(19)                               = 0
openat(AT_FDCWD, "/usr/bin/../libexec/opentrack", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 19
fstat(19, {st_mode=S_IFDIR|0755, st_size=3488, ...}) = 0
getdents64(19, 0x559f5923ac30 /* 37 entries */, 32768) = 1608
getdents64(19, 0x559f5923ac30 /* 0 entries */, 32768) = 0
close(19)                               = 0
openat(AT_FDCWD, "/usr/bin/../libexec/opentrack", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 19
fstat(19, {st_mode=S_IFDIR|0755, st_size=3488, ...}) = 0
getdents64(19, 0x559f5923ac30 /* 37 entries */, 32768) = 1608
getdents64(19, 0x559f5923ac30 /* 0 entries */, 32768) = 0
close(19)                               = 0
openat(AT_FDCWD, "/usr/bin/../libexec/opentrack", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 19
fstat(19, {st_mode=S_IFDIR|0755, st_size=3488, ...}) = 0
getdents64(19, 0x559f5923ac30 /* 37 entries */, 32768) = 1608
getdents64(19, 0x559f5923ac30 /* 0 entries */, 32768) = 0
close(19)                               = 0
```

sthalik, 2025-08-10:
Can you get a stack trace?

sthalik, 2025-08-10:
Try running [fix-keybindings-broken-linux](https://github.com/opentrack/opentrack/commits/fix-keybindings-broken-linux/).

SeanConn15, 2025-08-11:
@sthalik that branch fixed the issue for me! thanks

ProfScrewloose, 2025-08-11:
Confirmed the fix-keybindings-broken-linux branch fixes the issue on Fedora 42.

commit 8e2f9ebd (2025-08-13): fix keybindings crash on linux

Fixes: #2059