opentrack/opentrack #2061: QPixmap: Must construct a QGuiApplication before a QPixmap

sornig, 2025-08-08:
I successfully built the latest master branch on Void Linux. However, when I try to launch the application, I get the following error:

```
QPixmap: Must construct a QGuiApplication before a QPixmap
Abort
```

sornig, 2025-08-08:
I left all options at their default, except I installed/enabled ONNXRuntime (linux-x64-1.22.0). I think this is a Qt6 issue, because the last time I built it was back in October (still on Qt5) and that ran fine.

sornig, 2025-08-10:
```
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff60a524f in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:89
#2  0x00007ffff6055fe2 in __GI_raise (sig=sig@entry=6) at raise.c:26
#3  0x00007ffff603eefc in __GI_abort () at abort.c:73
#4  0x00007fffcc292113 in QMessageLogger::fatal(char const*, ...) const () from /usr/lib/libQt5Core.so.5
#5  0x00007fffcc8e9a84 in ?? () from /usr/lib/libQt5Gui.so.5
#6  0x00007fffcc9ca064 in QPixmap::QPixmap() () from /usr/lib/libQt5Gui.so.5
#7  0x00007fffcc9d82db in QPixmapIconEngine::addFile(QString const&, QSize const&, QIcon::Mode, QIcon::State) () from /usr/lib/libQt5Gui.so.5
#8  0x00007fffcc9da75a in QIcon::addFile(QString const&, QSize const&, QIcon::Mode, QIcon::State) () from /usr/lib/libQt5Gui.so.5
#9  0x00007fffcc9dac43 in QIcon::QIcon(QString const&) () from /usr/lib/libQt5Gui.so.5
#10 0x00007fffd4020e6e in non-virtual thunk to wine_metadata::icon() () at ftnoir_protocol_wine.h:104
#11 0x00007ffff7f121bc in dylib::dylib (this=0x55555572e700, filename_=..., t=dylib_type::Protocol, load_mode=dylib_load_norm) at plugin-support.hpp:93
#12 0x00007ffff7f12c56 in std::_Construct<dylib, QString, dylib_type const&, dylib_load_mode&> (__p=0x55555572e700) at stl_construct.h:109
#13 std::allocator_traits<std::allocator<void> >::construct<dylib, QString, dylib_type const&, dylib_load_mode&> (__p=0x55555572e700) at alloc_traits.h:706
#14 std::_Sp_counted_ptr_inplace<dylib, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<QString, dylib_type const&, dylib_load_mode&> (this=0x55555572e6f0, __a=...)
    at shared_ptr_base.h:607
#15 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<dylib, std::allocator<void>, QString, dylib_type const&, dylib_load_mode&> (__a=..., this=<synthetic pointer>,
    __p=<synthetic pointer>: <optimized out>) at shared_ptr_base.h:969
#16 std::__shared_ptr<dylib, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<void>, QString, dylib_type const&, dylib_load_mode&> (this=<synthetic pointer>, __tag=...) at shared_ptr_base.h:1713
#17 std::shared_ptr<dylib>::shared_ptr<std::allocator<void>, QString, dylib_type const&, dylib_load_mode&> (this=<synthetic pointer>, __tag=...) at shared_ptr.h:463
#18 std::make_shared<dylib, QString, dylib_type const&, dylib_load_mode&> () at shared_ptr.h:1008
#19 Modules::enum_libraries (library_path=..., load_mode=load_mode@entry=dylib_load_norm) at plugin-support.hpp:237
#20 0x00007ffff7f0eb70 in Modules::Modules (this=0x555555b01268, library_path=..., load_mode=dylib_load_norm) at plugin-support.hpp:179
#21 State::State (this=this@entry=0x555555b01268, library_path=...) at state.cpp:24
#22 0x0000555555569448 in main_window::main_window (this=this@entry=0x555555b01240) at qstring.h:1437
#23 0x0000555555571e40 in std::make_unique<main_window> () at unique_ptr.h:1077
#24 operator() (__closure=<optimized out>) at main.cpp:14
#25 std::__invoke_impl<std::unique_ptr<main_window, std::default_delete<main_window> >, main(int, char**)::<lambda()>&> (__f=...) at invoke.h:61
#26 std::__invoke_r<std::unique_ptr<QWidget>, main(int, char**)::<lambda()>&> (__fn=...) at invoke.h:114
#27 std::_Function_handler<std::unique_ptr<QWidget, std::default_delete<QWidget> >(), main(int, char**)::<lambda()> >::_M_invoke(const std::_Any_data &) (__functor=...) at std_function.h:291
#28 0x00007ffff7f8242f in std::function<std::unique_ptr<QWidget, std::default_delete<QWidget> >()>::operator() (this=0x7fffffffd930) at std_function.h:591
#29 otr_main (argc=<optimized out>, argv=<optimized out>, make_main_window=...) at init.cpp:345
#30 0x0000555555560526 in run_application<main(int, char**)::<lambda()> > (fun=..., argc=<optimized out>, argv=<optimized out>) at init.hpp:14
#31 main (argc=<optimized out>, argv=<optimized out>) at main.cpp:14
```

sthalik, 2025-08-10:
Are you sure you aren't linking two separate versions of Qt 6? Try using `LD_DEBUG`.