opentrack/opentrack #2064: PS3 Eye - no fps option above 60?

billybong, 2025-08-15:
When following the instructions for installing the PS3 Eye camera drivers I can only select between the default, 30 and 60 fps alternatives. Screenshots seem to hint that 120 would be possible, but I can’t see it.
Is this a known issue and should I simply leave fps option at default value to achieve 120?

Otherwise, is there some way to debug how many fps I’m actually squeezing through the camera?

Kudos and thanks for this awesome project.

Len62, 2025-08-16:
Eye camera: up to 75 fps at 640×480 resolution and up to 187 fps at 320×240 resolution. At 640x480 I get between 50 and 74 FPS. With OT running you can open the input settings and see the FPS at the bottom of the screen.

[image]

billybong, 2025-08-16:
Hmm, I can see you could select the 75 Hz in the dropdown. Mine is either 30fps or 60. Which drivers are you using?

I also have an internal camera since I’m using a laptop. Maybe OT gets confused by that?

Len62, 2025-08-16:
It's not a dropdown, it's a scroll bar. I just use the up/down arrows to change the FPS. I use the libusbk driver:

[image]

Len62, 2025-08-17:
Ok, I using the latest test version. Link: https://ananke.misaki.pl/opentrack/opentrack-test-20250729_01.7z

billybong, 2025-08-17:
> Ok, I using the latest test version. Link: https://ananke.misaki.pl/opentrack/opentrack-test-20250729_01.7z

Oh! That fixed it for sure 👍 
Bleeding edge it is then.

closed 2025-08-17 by billybong