opentrack/opentrack #2082: Opentrack suddenly freezes while playing DCS

virus261, 2025-09-21:
Hi, I've been using opentrack for about a year now. I never had issues until 2 months ago. The issue is opentrack suddenly stops responding and freezes during gameplay. The freeze can take several minutes and comes back to life eventually but it freezes again. Doesn't even last 20 minutes to freeze again. After having this issue I have updated to opentrack 2024.1.1 and also updated the ps3 camera drivers to libusK version but the issue persists. I am using IR leds for tracking. How to fix this issue?

if you need additional information please ask

sthalik, 2025-12-03:
There shouldn't be any opentrack freezes at all. See if you can send a process minidump using Task Manager on `opentrack.exe` once it's hung. Then send it here along with the opentrack version you're using.

virus261, 2025-12-10:
Here are 3 instances that tracking was stopped but the app itself was responsive. I was able to press stop and restart the tracking without app crashing or going hung state. Normally it would go unresponsive once I click stop but for these instances it didn't happen.  App version is 2024.1.1

virus261, 2025-12-16:
Upon further investigation I've noticed that whenever the situation mentioned above happens it also disconnects my headphones. I have to unplug then plug it again. 

After the issue I've removed the drivers for the ps3 camera (libusK) and removed the device. Then reinstalled the drivers using libusK again and reinstalled ps3eyesubprocess_v1.0.27-31ge678b3fa as mentioned in a other post. For now it looks like fixed and haven't had any issue whatsoever for 2 hours or so.

@Henparty Try doing these steps and see if it works for you too.

https://github.com/opentrack/opentrack/issues/1877 ps3eye subprocess link
https://github.com/opentrack/opentrack/wiki/PS3-Eye-open-driver-instructions ps3eye libusK driver link

Henparty, 2025-12-17:
Thanks, will keep trying with the logicam and as many drivers cleaned out as I can.

virus261, 2026-01-17:
Do you have any updates Mr @sthalik? I still have this issue after 2026.01.01 release with the libusb drivers.