opentrack/opentrack #2083: Open track stop working after Windows 24H2 update.

Joeblack1984, 2025-09-26:
I am using Beam Eye Tracker and open track together for a while.. When Windows update to 24H2 version open track stop working. I changed nothing. When i start open track it closed itself.. Is there any solution for this ?

Joeblack1984, 2025-09-27:
Non of the opentrack builds working for me. I will turn back to windows 23H2 version.

kakra, 2025-10-03:
A friend of mine has likely the same or similar problem with the Windows version:

opentrack does not quit but can no longer record from the camera when using NeuralNet. For other inputs that use the camera, the camera still shows but of course, the octopus doesn't move.

TomCrum-wdm, 2025-10-28:
Thank you for your help😊.It seem working but I want a function just like Beam Eye Tracker,which create a circle on screen instead of changing cursor position .Change cursor position is so awful because eye track can not be precise.

[image]